G0WCZ/cwc #32: Add startup tone

m0vva, 2019-05-08:
It would be good to get some audible and/or visual feedback when cwc-station has started.  Maybe send out an "R" and flash the same on the LED?

grasuth, 2019-05-08:
Yes!
Also with the slightly updated keepalive, the station gets a response from the reflector, so can tell the reflector is there and connected.  Maybe a connected LED.  or some cw..  "dit-dit dit" maybe (the very old landline morese "is the frequency in use" query/response)

are there two states here:  1/ running/startup . 2/ connected?

m0vva, 2019-05-08:
Good idea on the connected LED.  I have already added a "status" LED on my Pi hat in preparation!  The LED could be off when there is no connection, flash whilst connecting and be solid when connected.

grasuth, 2019-06-12:
adding a sepate issue for better LED dynamics (breathing, pulsing stuff)

closed 2019-06-12 by grasuth